jgillick/arduino-LEDFader #6: Modifying multiple fader example

relishgargler, 2015-04-22:
I'm trying to modify your example for randomly fading 6 LEDs to work with 11. If I just change the 
`#define LED_NUM 6` line to 
`#define LED_NUM 11, `
and then add in the additional pins under the 
`LEDFader leds[LED_NUM] = { `
section, it fades the 6 LEDs you originally chose, but the others will only flash on or off. 

Am I missing something? I don't see anything in the rest of the code that would exclude the other pins from fading, but I'm not 100% sure what I'm doing yet.

jgillick, 2015-05-07:
Sorry for the delay, I was out of the country on vacation. I'm not sure why this is not working as expected from those little snippets. Are you sure the pins you're plugging the other LEDs to support PWM output? Otherwise, please put your entire sketch code in this ticket and we can take it from there.

relishgargler, 2015-05-13:
That's exactly what was wrong. I'd only had an arduino for about a day and didn't realize only specific pins had PWM support.
I did notice that I had to change led.fade in your sample to led->fade in order to get the code to compile.

closed 2015-05-13 by relishgargler

jgillick, 2015-05-26:
Glad that got it working!