jgillick/arduino-LEDFader #7: Maximum fading time

Ivanooe, 2015-05-15:
Hello 

thank a lot for this library, is very helpful.
I have only a question...I saw into the library that you define the interval (time of fading) like an unsigned int.
In this way the maximum time that I can set up it's about 65 second.
I need an interval like 1 hour... Is possible to change the interval from unsigned int to unsigned long?
Or you have choose this type of variable for a specific reason?
Thanks in advance.

Ivan

jgillick, 2015-05-18:
Hi Ivan,

The decision was purely because, for most needs, I didn't expect the interval to be longer than 65 seconds. I don't think it should be a problem for you to change it to a long data type. If you do this, let me know how it works!

Ivanooe, 2015-05-20:
I tried to do this changement...
I modify all the variables from unsigned int to unsigned long, but unluckly I didn't see any improve.

I tried to set up the interval over 70000 and more, but I can't see any difference...It works, but the fade time seems not change.

I don't know....

jgillick, 2015-05-26:
That's odd. How long does it seem to fade?

r00tGER, 2016-01-27:
Another way is dividing millis() to 1000. In this case precision will be in seconds, and max time ~18 hr.

electroc, 2016-05-10:
hi all, and thanks jgillick for this great library

the unsigned long method didnt work for me either, but i eventually figured r00tGER dividing the millis();

i made my fader to run up to 600 seconds by dividing all the millis(); in the .cpp file

`LINE#108  last_step_time = millis()/10;`
`LINE#140  unsigned long now = millis()/10;`
`LINE#167  last_step_time = millis()/10;`

and dividing the MIN_INTERVAL in the .h file as well (default was 20):

`LINE#15 #define MIN_INTERVAL 2`

now the FADE_TIME is in milliseconds*10 so 2 seconds are now 200 and not 2000 , just to keep in mind

GavC666, 2018-02-04:
Hi
Changing around half a dozen or so unsigned int's to unsigned long within c++ file allows the user to then increase interval time. Worked no problem for me (noob). 
Thank you jgillick for this library, I have searched a long time and this is ideal for simulating sunrise and sunset in my snake Viv.
Kind Regards
Gav

jgillick, 2018-02-04:
@GavC666 that makes sense. Mind submitting a pull request with those changes and I'll merge it into the library?