jgillick/arduino-LEDFader #14: Use function in "Curve::exponential" instead of table

sphh, 2016-09-18:
Hi,

I like your library very much. Thanks for your work!

You notice in `Curve.cpp`:

    // If the compiler could build the above table for us, we could have "simply":
    //   return !i ? 0 : round(exp(log(MAXOUTPUT) * i / MAXINPUT));

I tried it and strangely enough the compiler does not complain, even if these functions are not mentioned in the Arduino reference, But executing this statement gives complete rubbish. Luckily we can replace the functions with their basic equivalents (assuming `MAXOUTPUT=MAXINPUT=255`):

- `log(255)/255` = `0.0217304452751311`
- `exp(x)` = `pow(2.71828182845905, x)`
- `round(x)` = `int(x + 0.5)`

Now we can use

    return !i ? 0 : int(pow(2.71828182845905, i * 0.0217304452751311) + 0.5);

and can get rid of that table!

jgillick, 2016-09-21:
The table is a performance enhancement so the calculation does not need to run for every frame of the fade. Performing those sorts of calculations can be expensive on an 8-bit AVR, especially if it's doing many other things. 

I believe the comment about the compiler complaining was meant to say " If the compiler **couldn't** build..."

@cinderblock can also shed some light on this.

sphh, 2016-09-21:
Sorry, that must have been a misunderstanding on my side. I thought because the table takes such a large space in the program memory, it would be preferable to calculate that value...

closed 2016-09-21 by sphh

cinderblock, 2016-10-05:
I could have used better wording in that description.

It was my intention that you could uncomment that line and use it with the same results. It makes it easier to make sure the table that is generated/used is correct.

In our case, the speed of the computation is assumed to be the most important thing since we're trying to output new values many times per second. For small displays, AVRs are already plenty fast. But this is important for bigger displays or where you're doing other things on your AVR. You are correct that this table uses a bunch of program space but this was the choice we made to make the computation faster.

My point about "if the compiler could..." is referring to if there was a way to tell the compiler: here is the function I want, make the lookup table for me for all possible input values.