jgillick/arduino-LEDFader #15: Does not compile for ESP8266 b/c of pgmspace.h

myhd, 2016-09-21:
A great little library, thank you!

**Can you make it also build for the fantastic ESP8266 platform?** 

Here is what I get when I try to compile for the [Arduino core for ESP8266 WiFi chip](https://github.com/esp8266/Arduino):

On Arduino 1.6.11 and [esp8266 board package](https://github.com/esp8266/Arduino), version 2.3.0: 

```
In file included from /Users/foo/Arduino/libraries/LEDfader/Curve.cpp:8:0:
/Users/foo/Documents/Arduino/libraries/LEDfader/Curve.h:11:26:
fatal error: avr/pgmspace.h: No such file or directory
 #include <avr/pgmspace.h>
                          ^
compilation terminated.
exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

When I just remove `Curve.cpp` and `Curve.h`, it all builds well. Perhaps related: [Use function in "Curve::exponential" instead of table](https://github.com/jgillick/arduino-LEDFader/issues/14). Thank you!

jgillick, 2016-09-21:
I don't have an ESP8266 to play with (although I keep intending to order one), so I'm not sure if this will work. It looks like there are 2 options:

 1) There's a version of `pgmspace.h` in the [ESP8266 repo](https://github.com/esp8266/Arduino/blob/master/cores/esp8266/pgmspace.h). Try replacing `<avr/pgmspace.h>` with `<pgmspace.h>` and see if it works. If it does, I'll update the code in the repo.

2) Alternatively, the ESP8266 is quite a bit more powerful than the standard 8-bit Arduino, so you can probably remove the table all together (like you're suggesting in the other ticket) and replace it with the math expression. The downside to this is that the math would need to be calculated for every frame of the animation, which isn't very clean or efficient. To integrate this into the library, we'd want to isolate that to the ESP8266 with an `#ifdef ARDUINO_ESP8266_RELEASE ` preprocessor block.

myhd, 2016-09-21:
Hah! That was easy. Thank you for the super fast reply. I tried your first suggestion, replacing `<avr/pgmspace.h>`with `<pgmspace.h>`. Compiles now! Looking forward to the updated repo.

By the way, perhaps I have overlooked it in the docs, but what exactly does Curve.h/.cpp do? How would one use it?

jgillick, 2016-09-23:
Curve.h can define custom LED fading progressions. The default is to do a linear fade, where each millisecond it changes the brightness by `x`. 

The problem with a linear fade is that, visually, it doesn't end up looking very linear and constant. ([this post explains it well](https://diarmuid.ie/blog/pwm-exponential-led-fading-on-arduino-or-other-platforms/)) So your code can define `exponential` as the fade curve, to flatten this out and make it look more natural. You'd use this curve like this:

```cpp
#include <LEDFader.h>
#include <Curve.h>

#define LED_PIN 3

LEDFader led;

void setup() {
  led  = LEDFader(LED_PIN);
  led.set_curve(&Curve::exponential);
}
```

The curve functions are really simple maps, where you pass a number between 0 - 255 and the function returns the actual color that should be represented at that position.

JamesTM5, 2016-11-08:
This is really helpful thanks, I am trying to get this to work on ESP8266 as well and hitting the same error.  I shall give your proposed fix a try.  In addition, I was trying to figure out how I would go about smoothing the fade as you describe and it is great to know you already have this built in!

pacav69, 2017-05-10:
another workaround is to replace this line

    #include <avr/pgmspace.h>

with the following:

    #if (defined(__AVR__))
    #include <avr/pgmspace.h>
     #else
    #include <pgmspace.h>

mahyarPourjabbar, 2017-12-09:
An update of Arduino AVR Board will entirely solve the problem.

[image]

closed 2019-11-18 by jgillick