jgillick/arduino-LEDFader #17: lights show

cristi134, 2017-03-11:
Hello, i'm trying to make an elaborate lights show using the first example this but i'm facing some problems:
1.i can't get it to loop. simply copying the part from void setup to void loop does not seem to work.
2.There is only one fade in your example, from 0 to 255 in 3 seconds. Simply adding more fades after this does not work either. How can i add more of them?
3.Can i use 'for' statement to create fade pattterns and repeat them (or maybe even create a function of a pattern and insert it in different places?)
Thanks in advance!

jgillick, 2017-03-14:
Unfortunately, this description doesn't give me enough information. Please tell me about your setup and show me the exact code you're using. What pin is the LED connected to?

 Per your second and third point, I think you'll find that my other example shows how to do this:
https://github.com/jgillick/arduino-LEDFader/blob/master/examples/multipleLEDs/multipleLEDs.ino

cristi134, 2017-03-15:
I only need 1 output pin. 
Below, a small sample of the code, with delay. I'm trying to achieve the same result but to be able to receive inputs at the same time. There will be 9 different light shows like the one in the example (but much longer), each of them triggered by an input pin.

int ledPin = 9;

void setup() {
  // put your setup code here, to run once:

}

void loop() {

  analogWrite(ledPin, 255); 
delay(5000);
  analogWrite(ledPin, 210); 
delay(1948);
  analogWrite(ledPin, 180); 
delay(3163);
  analogWrite(ledPin, 150); 
delay(5000); 
  analogWrite(ledPin, 180); 
  delay(5100); 
      for (int fadeValue = 180 ; fadeValue >= 50; fadeValue -= 1) {
    analogWrite(ledPin, fadeValue);
    delay(33);
      }
         for (int fadeValue = 50 ; fadeValue <= 255; fadeValue += 1) {
    analogWrite(ledPin, fadeValue);
    delay(33);
      }  
            for (int fadeValue = 255 ; fadeValue >= 60; fadeValue -= 1) {
    analogWrite(ledPin, fadeValue);
    delay(20);
            }
  for (int i = 0; i < 21; i++)
{
  for (int fadeValue = 60 ; fadeValue <= 85; fadeValue += 1) {
    analogWrite(ledPin, fadeValue);
    delay(5);
  }
              for (int fadeValue = 85 ; fadeValue >= 60; fadeValue -= 1) {
    analogWrite(ledPin, fadeValue);
    delay(5);
              }
    
}
}

jgillick, 2017-03-15:
Can you point me to the example you're taking this from? This doesn't look like any code out of my examples.

To your question, anytime you add a `delay`, it'll stop your entire program until that delay completes.

cristi134, 2017-03-15:
I made the above example, it's not from your code. I'm trying to 'translate' it in your code (to get rid of the delay) but i've run into 3 major problems, mentioned in my first post.

jgillick, 2017-03-15:
From what I can see in your code, there's nothing obvious that wouldn't make it loop. Although, it would be easier to read if the intents used were consistent (it looks like it's a mix of tabs and spaces which is messing up the visual hierarchy of the code)

To your original question. You should be able to setup multiple light patterns using for loops without delays. What you want to do is make the light changes based on the time, not delays, in order to do this. That is basically how my library works. Here's more information on this:
https://www.arduino.cc/en/Tutorial/BlinkWithoutDelay

closed 2019-11-18 by jgillick