jgillick/arduino-LEDFader #18: 2 leds

feelgoodgirl, 2017-03-23:
Heyy,

First, I want to give compliments for this library. For one LED does that really super good.

But now I have a little problem. For a program I would like to use the "fading library." The goal is to, for example, push button C and LED C starts fading. And when i push button B, LED B starts fading. I want this, however, connect to different pins. Do you happen to have a program that does this? The fact I am unable to change your preview of several LEDs.

I look forward to your reaction,

greetings yvette

feelgoodgirl, 2017-03-23:
Now i use to following:

`#include <LEDFader.h>

LEDFader led = 10;
LEDFader led2 = 13;
int button = 12;
int button2 = 5;
int fading = 0;
LEDFader A;

void setup() {
  pinMode (button, INPUT);
  pinMode (button2, INPUT);
  //LEDFader(led);
}

void loop() {
  if (digitalRead(button) == HIGH) {
    A = LEDFader (led);
    fading = 1;
  }

  if (digitalRead(button2) == HIGH) {
    A = LEDFader (led2);
    fading = 1;
  }

  if (fading == 1)  {
    A.update();

    if (A.is_fading() == false) {
      if (A.get_value() == 255) {
        A.fade(50, 2000);
      }
      else {
        A.fade(255, 1000);
      }
    }
  }
}`

if you had some idea how to make it better, let me know

jgillick, 2017-03-23:
Thank you for posting your code. I have a few ideas here. Why are you connecting the fader for both pins to the same variable? This will likely cause a problem with finishing the first fade. I would also put `A.update();` at the top of the `loop` function. I notice that `fading` never goes to `0`, is this intended? Also, you should probably begin the fade when the button is pressed.

Have you had a look at the Multiple LEDs example:
https://github.com/jgillick/arduino-LEDFader/blob/master/examples/multipleLEDs/multipleLEDs.ino

I would probably try to code the solution like this (I have not compiled or tried this code):

```javascript
#include <LEDFader.h>

LEDFader led1 = 10;
LEDFader led2 = 13;
int button = 12;
int button2 = 5;
int fading = 0;

LEDFader faderA;
LEDFader faderB;

void setup() {
  pinMode (button, INPUT);
  pinMode (button2, INPUT);

  faderA = LEDFader (led1);
  faderB = LEDFader (led2);
}

void loop() {
  faderA.update();
  faderB.update();

  if (digitalRead(button) == HIGH) {
    faderA.fade(255, 1000);
  }
  if (digitalRead(button2) == HIGH) {
    faderB.fade(255, 1000);
  }

  // Fade out
  if (faderA.is_fading() == false && faderA.get_value() == 255) {
    faderA.fade(50, 2000);
  }
  if (faderB.is_fading() == false && faderB.get_value() == 255) {
    faderB.fade(50, 2000);
  }
}
```

feelgoodgirl, 2017-03-24:
Thank you for your comment and for your code. I watched him for a moment and recreated it. This corresponds roughly to what I mean. Only now does it only fade ones, but here I can do something with.

There is no special reason why they do not to '0', personally i liked it better if it does not completely go out.

The reason I linked two pins on one variable is because I really want a "fading" program use for all my LEDs.
In the program you have now, every LED use its own program and if you are going to use more LEDs it only get more.
I tried to understand your Multiple LEDs example. But without succes. I am a beginner with Arduino.

Can you explain to me why you need this: 'faderA.is_fading () == false'?

jgillick, 2017-03-24:
Each fade variable controls the fade for a single LED. This is why you need a new one for every LED you wish to control.  It cannot be shared with multiple LEDs. (my second code example in this message might be what you were looking for, though)

`faderA.is_fading () == false` Is checking to see if the current fade (set by `faderA.fade( ... )`) has completed. In your case, I'm assuming you do not want to fade back down until it completes fading up to 255.

You're correct that the example I provided will only fade the LED once. If you want it to continue fading up and down after pressing the button, you would modify the loop function to be like this:

```cpp
void loop() {
  faderA.update();
  faderB.update();

  // Fade up when button is pressed
  if (digitalRead(button) == HIGH) {
    faderA.fade(255, 1000);
  }
  if (digitalRead(button2) == HIGH) {
    faderB.fade(255, 1000);
  }

  // Fade out automatically
  if (faderA.is_fading() == false) {
    if (faderA.get_value() == 255) {
      faderA.fade(50, 2000);
    }
    else {
      faderA.fade(255, 2000);
    }
  }
  if (faderB.is_fading() == false) {
    if (faderB.get_value() == 255) {
      faderB.fade(50, 2000);
    }
    else {
      faderB.fade(255, 2000);
    }
  }
}
```

Also, if each LED will be doing the same thing when you press the button, and you don't want to repeat the same code for each LED, you can create a generic function that takes the fader as an argument. This example is using something in C programming called pointers:

```cpp
#include <LEDFader.h>

LEDFader led1 = 10;
LEDFader led2 = 13;
int button1 = 12;
int button2 = 5;

LEDFader faderA;
LEDFader faderB;

void setup() {
  pinMode (button1, INPUT);
  pinMode (button2, INPUT);

  faderA = LEDFader (led1);
  faderB = LEDFader (led2);
}

void loop() {
  updateFadeState(&faderA, button1);
  updateFadeState(&faderB, button2);
}

void updateFadeState(*fader, btn) {
  fader->update();

  if (digitalRead(btn) == HIGH) {
    fader->fade(255, 1000);
  }

  if (fader->is_fading() == false) {
    if (fader->get_value() == 255) {
      fader->fade(50, 2000);
    }
    else {
      fader->fade(255, 2000);
    }
  }
}
```

closed 2019-11-18 by jgillick