jgillick/arduino-LEDFader #21: Startup Fade in Setup()

RossAWaddell, 2017-06-26:
First of all, thank you so much for this library! A fellow hobbyest pointed me to this code and with it I've been able to create the exact look I've been after for years, with only one small issue:

I want to fade up an LED in the void setup() loop over 2-3 seconds, but no matter how long I make the duration (it's set below to 7 seconds) it just sorta turns on - no real gradual fading. I'm using an Arduino Pro Mini board (5V, 16Mhz) and all 6 PWM pins.

Here's my code:
`#include <LEDFader.h>              

// Config for blinking coloured LEDs
// =================================
#define LED_NUM 5                  // Number of LEDs

// 5 pin blinking LED setup
LEDFader leds[LED_NUM] = {
  LEDFader(3),                     // PWM pin
  LEDFader(5),                     // PWM pin
  LEDFader(6),                     // PWM pin
  LEDFader(9),                     // PWM pin
  LEDFader(10),                    // PWM pin
};

// Config for pulsing amber LEDs
// =============================
#define AMBER_LED_PIN 11           // Amber LEDs (5 in pentagon shape)
#define FADE_UP_TIME 7000          // Time to power up
#define PULSE_TIME 750             // Fade down/up pulse time
#define MAX_BRIGHTNESS 225         // Max brightness of LEDs
#define MIN_BRIGHTNESS 125         // Min brightness of LEDs to fade to (pulse)
#define DIR_UP 1
#define DIR_DOWN -1

LEDFader amber_led;
int direction = DIR_UP;

void setup() {
  randomSeed(analogRead(0));

  amber_led = LEDFader(AMBER_LED_PIN);
  amber_led.fade(MAX_BRIGHTNESS, FADE_UP_TIME);
}
 
void loop() {

  // Pulsing amber LEDs
  // ==================

  amber_led.update();

  // LED no longer fading, switch direction
  if (!amber_led.is_fading()) {

    // Fade down
    if (direction == DIR_UP) {
      amber_led.fade(MIN_BRIGHTNESS, PULSE_TIME);
      direction = DIR_DOWN;
    }
    // Fade up
    else {
      amber_led.fade(MAX_BRIGHTNESS, PULSE_TIME);
      direction = DIR_UP;
    }
  }

  // Blinking coloured LEDs
  // ======================
  
// try LEDFader's faster and slower methods:

// Increase the current fading speed by a number of milliseconds
//    void faster(int by_seconds);

// Decrease the current fading speed by a number of milliseconds
//    void slower(int by_seconds);

  // Update all LEDs and start new fades if any are done
  for (byte i = 0; i < LED_NUM; i++) {
    LEDFader *led = &leds[i];
    led->update();
 
    // This LED is not fading, start a new fade
    if (led->is_fading() == false) {
      int duration = random(125, 500);             // Duration between 100 to 750 ms (0.75 s)
      // int duration = random(100, 750);
      // int duration = random(0, 750);
      // int duration = random(1000, 3000);
 
      // Fade Up
      if (led->get_value() == 0) {
        byte intensity = random(75, 150);           // Brightness intensity between 0 (off) to 255 (maximum)
        // byte intensity = random(15, 50);
        // byte intensity = random(0,75);
        // byte intensity = random(100, 255);

        led->fade(intensity, duration);
      }
      // Fade Down
      else {
        led->fade(0, duration);
      }
    }
  }
}`

jgillick, 2017-06-27:
Hmmm. I'm not really sure. Your code looks right. A couple things to try:

Perhaps to explicitly set it to off initially:
```ino
amber_led = LEDFader(AMBER_LED_PIN);
amber_led.set_value(0);
amber_led.fade(MAX_BRIGHTNESS, FADE_UP_TIME);
```

Alternatively, try setting the fade time from 7 seconds to something like 2 seconds.

Let me know if either changes it.

RossAWaddell, 2017-06-27:
I simplified the code to just the LEDs I want to fade in:

#include <LEDFader.h>              // Don't forget to include the LEDFader libraries folder
#include <Curve.h>

// Config for pulsing amber LEDs
// =============================
#define AMBER_LED_PIN 11           // Amber LEDs (5 in pentagon shape)
#define FADE_UP_TIME 5000          // Time to power up
#define PULSE_TIME 600             // Fade down/up pulse time
#define MAX_BRIGHTNESS 235         // Max brightness of LEDs
#define MIN_BRIGHTNESS 180         // Min brightness of LEDs to fade to (pulse)
#define DIR_UP 1
#define DIR_DOWN -1

LEDFader amber_led;
int direction = DIR_UP;

void setup() {
  amber_led = LEDFader(AMBER_LED_PIN);
  amber_led.set_curve(&Curve::exponential);
  amber_led.set_value(0);
  amber_led.fade(MAX_BRIGHTNESS, FADE_UP_TIME);
  //amber_led.set_curve(&Curve::linear);
}
 
void loop() {

  // Pulsing amber LEDs
  // ==================

  amber_led.update();

  // LED no longer fading, switch direction
  if (amber_led.is_fading() == false) {

    // Fade down
    if (direction == DIR_UP) {
      amber_led.fade(MIN_BRIGHTNESS, PULSE_TIME);
      direction = DIR_DOWN;
    }
    // Fade up
    else {
      amber_led.fade(MAX_BRIGHTNESS, PULSE_TIME);
      direction = DIR_UP;
    }
  }
}

By adding the curve=exponential it looks better, but it's not exactly smooth. Here's a video: https://youtu.be/cxZGFXAEg3A <https://youtu.be/cxZGFXAEg3A>

RossAWaddell, 2024-05-25:
(apologies for re-opening this issue, but I'm trying to solve this again)

I tried adding amber_led.set_value(0); as you suggested, but no change - the LED goes straight on before fading up/down. I tried the example sketch, **singleLED**, and same result (see video).

I've got pin 3 of my Uno connected to the -ve lead of the LED, and then the +ve lead goes through a resistor to +5v.

https://github.com/jgillick/arduino-LEDFader/assets/29708900/2e24acd4-65d2-4db9-a7b8-fdc859b07edf

RossAWaddell, 2024-07-23:
Figured out why the FADE UP doesn't work - it's because the -ve pin of the LED is connected to the Arduino pin, not the +ve pin as usual when just using a single LED. The reason I'm using the -ve pin is that in my model I need to drive 10 LEDs so I go thru a ULN2803A transistor, hence that's how I'm breadboarding it when prototyping. If I keep this connection and change the code in your 'fade_on' sketch to use the FadeOff() function, it fades up properly.

So, the question becomes: if I need to use the -ve LED pin connection to the Arduino, how do I start with a FADE UP?