jgillick/arduino-LEDFader #24: does not seem compatible with Teensy

robocutstudio, 2017-09-26:
any plans on adding support ? ;)
really good btw !
thanks

jgillick, 2017-09-29:
It should work with the Teensy. What happens when you try to use it?

robocutstudio, 2017-09-29:
nothing :)
pin is not outputting anything..
when I use the same code on an arduino micro it works..

-- 

Philippe Savard
Fondateur / Directeur technique
514­-664­-3155 <https://goo.gl/maps/MRtZVQ16he32>
www.robocutstudio.com <http://robocutstudio.com/>
5425 ave Casgrain suite #704  <https://goo.gl/maps/MRtZVQ16he32>
Montréal, QC, H2T 1X6, Canada <https://goo.gl/maps/MRtZVQ16he32>
<https://www.facebook.com/robocutstudio/>
<https://www.instagram.com/robocutstudio/>
<https://twitter.com/robocutstudio>
<https://www.linkedin.com/company/2145378?trk=tyah&trkInfo=tarId%3A1413287409713%2Ctas%3Arobocut%2Cidx%3A1-1-1>

jgillick, 2017-10-01:
Please provide more information. What pin do you have it connected to? Is it a PWM pin? You might have better results if you also add a 20k pull-down resistor to that pin (probably shouldn't be needed, though).

robocutstudio, 2017-10-04:
ok I will test soon and let you know :)

-- 

Philippe Savard
Fondateur / Directeur technique
514­-664­-3155 <https://goo.gl/maps/MRtZVQ16he32>
www.robocutstudio.com <http://robocutstudio.com/>
5425 ave Casgrain suite #704  <https://goo.gl/maps/MRtZVQ16he32>
Montréal, QC, H2T 1X6, Canada <https://goo.gl/maps/MRtZVQ16he32>
<https://www.facebook.com/robocutstudio/>
<https://www.instagram.com/robocutstudio/>
<https://twitter.com/robocutstudio>
<https://www.linkedin.com/company/2145378?trk=tyah&trkInfo=tarId%3A1413287409713%2Ctas%3Arobocut%2Cidx%3A1-1-1>

closed 2019-11-18 by jgillick