jgillick/arduino-LEDFader #28: Pin 0 on A-Star 328PB Micro Doesn't Work

RossAWaddell, 2019-11-16:
Is there anything special about pin 0 in regards to your library? I can use this successfully on other pins, just not this one on the A-Star 328PB Micro. I looked through the code but couldn't see anything that would make it not work. I can use analogWrite() on this pin and other LED fading libraries, just not this one (which I love and have used extensively in my scale model work). I confirmed pin 0 doesn't work with the sindleLED example:

```
#include <LEDFader.h>

/*
  Fades a single LED up and down using LED Fader.
  Connect an LED to pin 3
*/

#define LED_PIN 0

#define FADE_TIME 2000

#define DIR_UP 1
#define DIR_DOWN -1

LEDFader led;
int direction = DIR_UP;

void setup() {
  led = LEDFader(LED_PIN);
  led.fade(255, FADE_TIME);
}

void loop() {
  led.update();

  // LED no longer fading, switch direction
  if (!led.is_fading()) {

    // Fade down
    if (direction == DIR_UP) {
      led.fade(0, FADE_TIME);
      direction = DIR_DOWN;
    }
    // Fade up
    else {
      led.fade(255, FADE_TIME);
      direction = DIR_UP;
    }
  }
}
```

jgillick, 2019-11-18:
This is an odd one. What happens when you try to use it? I assume the LED doesn't come on at all?

Here's my theory: The library doesn't set the pin output mode itself and `D0` is by default `RX`, which would be an input pin. Try this line in the `setup()` method:

```c++
pinMode(0, OUTPUT);
```

I don't think it has anything to do with the line you referenced in the header. That's just a default initialization and if you pass the pin number explicitly, that will be ignored.

**Random thoughts**

This library shouldn't do anything which would affect pin `D0` from working. On the [arduino UNO](https://www.arduino.cc/reference/en/language/functions/analog-io/analogwrite/), the `D0` pin does not have PWM output. However, on the [A-Star 328PB Micro](https://www.pololu.com/docs/0J74/3) it appears to have PWM output on `D0` via the OC3A PWM output.

RossAWaddell, 2019-11-18:
Unfortunately, that didn't work. I even added:

`UCSR0B = 0;`

but that didn't help, either. And yes, the LED doesn't come on at all.

jgillick, 2019-11-21:
That's really odd. Offhand I'm not sure why that would happen. I'll see if I have any Atmega 328PB chips lying around I can try this on.

jgillick, 2019-11-21:
Just out of curiosity, does `led.set_value(125);` set the LED to half brightness?

RossAWaddell, 2019-11-22:
Same result - the LED does not light.

```
#define LED_PIN 0
LEDFader led;

void setup() {
  led = LEDFader(LED_PIN);
  led.set_value(125);
}

void loop() {
  led.update();
}
```

jgillick, 2019-12-18:
Ah ha! I think I found the problem: 
https://github.com/jgillick/arduino-LEDFader/blob/master/LEDFader.cpp#L30

The code is first checking if the pin has been set. Well pin `0` is equivalently `false`. You can remove any of those lines. I'll try to update the library soon, but would also take a pull request if someone else wanted to take a stab at it. The library needs to be initialized with a pin, so those checks are likely moot.

RossAWaddell, 2019-12-19:
That's definitely the issue - if I comment out:

```
   // No pin defined
  //if (!pin) {
  //    return;
  //}
```
then all works as expected.

I would happily do a pull request for this, but wanted to check with you first - is it sufficient to just remove those checks from all functions, or would it be better to initialize the class with a signed int instead, as in:

```
// Create a new LED Fader for a pin
    LEDFader(int pwm_pin=-1);
```
Then the checks can remain, but like this:

```
   // No pin defined
  if (pin == -1) {
      return;
  }
```

jgillick, 2019-12-20:
I think it's safe to remove those checks entirely. A pin must be defined at initialization, so the checks are really unnecessarily.

RossAWaddell, 2019-12-20:
Ok, I’ll remove them and submit a pull request.

RossAWaddell, 2019-12-21:
Done

jgillick, 2020-01-06:
Great. Did you submit a pull request?

RossAWaddell, 2020-01-06:
I did, although this is my first Github pull request. Do you not see it?

jgillick, 2020-01-16:
It would show up here:
https://github.com/jgillick/arduino-LEDFader/pulls

Can you send me a link to it?

RossAWaddell, 2020-01-17:
https://github.com/RossAWaddell/arduino-LEDFader

RossAWaddell, 2020-01-17:
It is showing up now under Pull requests.

jgillick, 2020-01-17:
Got it and merged! Thanks @RossAWaddell, I really appreciate the help on this one.

closed 2020-01-17 by jgillick